auto-pts/auto-pts #249: [MESH] Test fail - MESH/NODE/TNPT/BV-07-C

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0c

Failed Test Case:
MESH/NODE/TNPT/BV-07-C

Description:
Receive a Segmented Access Message - Unicast

Log:
[MESH_MESH_NODE_TNPT_BV-07-C.log](https://github.com/intel/auto-pts/files/3010392/MESH_MESH_NODE_TNPT_BV-07-C.log)

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Failed Test Case:
GAP/SEC/AUT/BV-14-C

Description:
Verify that the IUT properly rejects the service request when there is insufficient authentication and then completes service correctly with Lower Tester as peripheral role.

Log:
[GAP_GAP_SEC_AUT_BV-14-C.log](https://github.com/intel/auto-pts/files/3010421/GAP_GAP_SEC_AUT_BV-14-C.log)

sjanc, 2021-05-21:
I think this is passing now both on Zephyr and Mynewt.

closed 2021-05-21 by sjanc