auto-pts/auto-pts #262: [GAP] BTP_TIMEOUT - GAP/SEC/AUT/BV-18-C

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Test Case:
GAP/SEC/AUT/BV-18-C

Error:
BTP TIMEOUT

Description:
Verify that the IUT can  pair with a device who's I/O capabilities do not allow an authenticated pairing, after a service request has been denied with the error response "Insufficient Authentication"

Log:
[GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-39-10-137807.log](https://github.com/intel/auto-pts/files/3010525/GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-39-10-137807.log)
[GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-39-47-526518.log](https://github.com/intel/auto-pts/files/3010526/GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-39-47-526518.log)
[GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-40-25-234473.log](https://github.com/intel/auto-pts/files/3010527/GAP_GAP_SEC_AUT_BV-18-C_2019-03-25T20-40-25-234473.log)
[GAP_GAP_SEC_AUT_BV-18-C.log](https://github.com/intel/auto-pts/files/3010528/GAP_GAP_SEC_AUT_BV-18-C.log)

MariuszSkamra, 2019-04-04:
PTS issue: CASE0054768

MariuszSkamra, 2021-03-05:
the issue was resolved in PTS

closed 2021-03-05 by MariuszSkamra