auto-pts/auto-pts #263: [GAP] BTP_TIMEOUT - GAP/SEC/AUT/BV-20-C

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Test Case:
GAP/SEC/AUT/BV-20-C

Error:
BTP TIMEOUT

Description:
Verify that the IUT when bonded with a peer device tests the bond when receiving the error response "Insufficient Authentication", if link is unencrypted, before eventually removing the bond to perform a new pairing.

Log:
[GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-42-54-731084.log](https://github.com/intel/auto-pts/files/3010531/GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-42-54-731084.log)
[GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-44-11-126697.log](https://github.com/intel/auto-pts/files/3010532/GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-44-11-126697.log)
[GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-45-27-785983.log](https://github.com/intel/auto-pts/files/3010533/GAP_GAP_SEC_AUT_BV-20-C_2019-03-25T20-45-27-785983.log)
[GAP_GAP_SEC_AUT_BV-20-C.log](https://github.com/intel/auto-pts/files/3010534/GAP_GAP_SEC_AUT_BV-20-C.log)

MariuszSkamra, 2019-04-04:
PTS issue: CASE0054768,
The same as https://github.com/intel/auto-pts/issues/262

MariuszSkamra, 2021-03-05:
the issue was resolved in PTS

closed 2021-03-05 by MariuszSkamra