auto-pts/auto-pts #264: [GAP] BTP_TIMEOUT - GAP/IDLE/NAMP/BV-01-C

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Test Case:
GAP/IDLE/NAMP/BV-01-C

Error:
BTP TIMEOUT

Description:
Verify the IUT can perform the Name Discovery Procedure and retrieve the device name from a peer device.

Log:
[GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-11-55-446292.log](https://github.com/intel/auto-pts/files/3010538/GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-11-55-446292.log)
[GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-12-38-174768.log](https://github.com/intel/auto-pts/files/3010539/GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-12-38-174768.log)
[GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-13-16-404943.log](https://github.com/intel/auto-pts/files/3010540/GAP_GAP_IDLE_NAMP_BV-01-C_2019-03-25T20-13-16-404943.log)
[GAP_GAP_IDLE_NAMP_BV-01-C.log](https://github.com/intel/auto-pts/files/3010541/GAP_GAP_IDLE_NAMP_BV-01-C.log)

tedd-an, 2019-04-02:
It looks like PTS issue.

When the IUT is trying to discover the primary service UUID, the PTS delays the response up to 30 seconds which causes the disconnect between IUT and PTS.

Reported to PTS:
CASE0054706

kolodgrz, 2019-04-03:
I think that it might be worth to check if only Read Using Characteristic UUID will work for it. That is Name disvoery procedure.

tedd-an, 2019-04-04:
@kolodgrz 
That's a good idea and that makes more sense since GAP/IDLE/NAMP/BV-02-C test, which is the opposite test of this uses Read Using Characteristic UUID.

But it looks like Read Using Characteristic UUID is not implemented in Zephyr BTP tester.
Will work on it.

Thanks for the tip!

PR #304 (2019-04-24): btp: Add support Read Using Characteristic UUID
This patch adds support Read Using Characteristic UUID to read the
device name.

Fixes: #264
commit 13c0c0c8 (2019-04-24): btp: Add support Read Using Characteristic UUID

This patch adds support Read Using Characteristic UUID to read the
device name.

Fixes: #264
closed 2019-04-24 by tedd-an