auto-pts/auto-pts #286: [GAP] INCONC - GAP/CONN/UCON/BV-06-C

tedd-an, 2019-03-26:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Test Case:
GAP/CONN/UCON/BV-06-C

Error:
INCONC

Description:
Verify the IUT in the Undirected Connectable Mode using Resolvable Private Address can connect with another device performing the General Connection Establishment Procedure.

Log:
[GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-47-38-443197.log](https://github.com/intel/auto-pts/files/3010819/GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-47-38-443197.log)
[GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-48-41-050148.log](https://github.com/intel/auto-pts/files/3010820/GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-48-41-050148.log)
[GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-49-42-811004.log](https://github.com/intel/auto-pts/files/3010821/GAP_GAP_CONN_UCON_BV-06-C_2019-03-25T19-49-42-811004.log)
[GAP_GAP_CONN_UCON_BV-06-C.log](https://github.com/intel/auto-pts/files/3010822/GAP_GAP_CONN_UCON_BV-06-C.log)

MariuszSkamra, 2019-04-04:
PTS issue created CASE0054758, I will push the updated version of this test case to use wid handler instead of scripted test case steps.

tedd-an, 2019-05-29:
This started to pass last 10 runnings.
Close for now and reopen if it starts to fail again.

closed 2019-05-29 by tedd-an