auto-pts/auto-pts #296: [MESH] MESH/NODE/PROV/BV-09-C - PTS TIMEOUT

tedd-an, 2019-04-05:
PTS version: 7.4.1
Interop.PTSControl.dll - old version by reverting the patch 8d03da0

Test Case:
MESH/NODE/PROV/BV-09-C

Error:
PTS TIMEOUT

[MESH_MESH_NODE_PROV_BV-09-C.log](https://github.com/intel/auto-pts/files/3048982/MESH_MESH_NODE_PROV_BV-09-C.log)
[MESH_MESH_NODE_PROV_BV-09-C_2019-04-04T22-40-45-483196.log](https://github.com/intel/auto-pts/files/3048983/MESH_MESH_NODE_PROV_BV-09-C_2019-04-04T22-40-45-483196.log)
[MESH_MESH_NODE_PROV_BV-09-C_2019-04-04T22-43-32-998792.log](https://github.com/intel/auto-pts/files/3048984/MESH_MESH_NODE_PROV_BV-09-C_2019-04-04T22-43-32-998792.log)

michal-narajowski, 2019-06-05:
A was able to pass this test case after

michal-narajowski, 2019-06-05:
Actually, I was able to pass it with 200000 ms.

closed 2021-05-21 by sjanc