auto-pts/auto-pts #317: [PTS] PTS Protocol Viewer crashes after running 2 ~ 3 days

tedd-an, 2019-05-29:
PTS Protocol Viewer crashes after running 2 or 3 days of scheduled testing.
[image]

This causes the AutoPTS stops working and make it unreliable for long execution.

Reported PTS issue:
CASE0056436

gailzhang622, 2020-08-10:
@tedd-an 
I start to try to run auto-pts.  But I failed at the very first step, that is trying to run autoptsserver.py. 
I'm wondering if you could help me on these questions.

PTS is started automatically successfully. But 
self._pts = win32com.client.Dispatch('ProfileTuningSuite_6.PTSControlServer')

But it failed to get any API of PTSControll

such as the failure log, also failed when invoking self.pts.GetPTSVersion() method.

The failure log:
```
PS D:\work\code\auto-pts> python.exe .\autoptsserver.py                                                                                                                                           Local IP address: ('192.168.3.19', 'fe80::4a1:2e5e:28b:deee') DNS None
Local IP address: ('10.10.25.102', 'fe80::dcda:7108:b319:f671') DNS None
Starting PTS ...
Traceback (most recent call last):
  File ".\autoptsserver.py", line 127, in <module>
    main()
  File ".\autoptsserver.py", line 115, in main
    pts = PyPTSWithXmlRpcCallback()
  File ".\autoptsserver.py", line 52, in __init__
    ptscontrol.PyPTS.__init__(self)
  File "D:\work\code\auto-pts\ptscontrol.py", line 267, in __init__
    self.restart_pts()
  File "D:\work\code\auto-pts\ptscontrol.py", line 391, in restart_pts
    self.start_pts()
  File "D:\work\code\auto-pts\ptscontrol.py", line 436, in start_pts
    self._pts.SetControlClientLoggerCallback(self._com_logger)
  File "C:\Users\Gail\AppData\Local\Temp\gen_py\3.8\369AACA1-4397-3048-87C0-727666D465CFx0x7x6.py", line 111, in SetControlClientLoggerCallback
    return self._oleobj_.InvokeTypes(1, LCID, 1, (3, 0), ((12, 1),),logger
```
My envrRonment:
Windows 10 64bit
Python 3.8.5 32bit
Pywin32 32bit for 3.8
PTS 7.6.1 Build 4

I haven't got PTS dongle, so I run this without dongle. Just try to run autoptsserver.py.

Could you please help take a look?
Thanks.

sjanc, 2021-05-21:
I think we have workaround for this in place.

closed 2021-05-21 by sjanc