auto-pts/auto-pts #330: [GAP/GAT/BV-05-C] PTS fails to perform write to Device Name in encryption required

MariuszSkamra, 2019-07-17:
IUT requires encryption to perform write to Device Name characteristic. PTS test case ends with inconclusive verdict, because IUT responds Insufficient Encryption error on ATT Write Request.

As stated in Core 5.1 Vol 3. Part C 12.1 DEVICE NAME CHARACTERISTIC:
"Optionally writable, authentication and authorization
may be defined by a higher layer specification or be
implementation specific."

Proposed change:
PTS should start encryption when above error is received and retry to write characteristic on encrypted link.

CASE0057820

KKopyscinski, 2020-10-23:
The issue is still present in PTS 7.6.2. As Core 5.2 Vol 3, Part C 12.1 DEVICE NAME CHARACTERISTIC:

> The Device Name characteristic value may be writable. If writable,
authentication and authorization may be defined by a higher layer specification
or be implementation specific.

TSE 15950 was filed

sjanc, 2021-09-10:
test pass in latest run

closed 2021-09-10 by sjanc