auto-pts/auto-pts #392: [L2CAP] INCONC -  L2CAP/LE/REJ/BI-02-C

jsplyy, 2020-06-22:
PTS version: 7.6.0

tester firmware: build tester app for nrf52840 based on zephyr master branch
workspace: workspaces/zephyr/zephyr-v2.2.0/zephyr-v2.2.0.pqw6

Test Case:
L2CAP/LE/REJ/BI-02-C

This case could pass with  Zephyr V2.2.0 base, but could not pass with V2.3
**PTS received an unexpected error from the L2CA module.**
Detail PTS log attached.
@michal-narajowski @carlescufi  Do you know this issue and run this test recently?
Is there anything in workspace need to be updated to align with new zephyr base?

[L2CAP_LE_REJ_BI_02_C_2020_06_22_14_24_06.zip](https://github.com/intel/auto-pts/files/4811674/L2CAP_LE_REJ_BI_02_C_2020_06_22_14_24_06.zip)

carlescufi, 2020-06-22:
This would be one for @vudentz I think, since it's an L2CAP test.

kromancer, 2020-09-25:
@jsplyy I believe that this is a PTS defect, which is also present in v7.6.1.
Check: [CASE0069926](https://bluetooth.service-now.com/nav_to.do?uri=x_bsig_case_manage_case.do?sys_id=04886cc81b2b14100999a9bfbd4bcbd1%26sysparm_view=Self_Service_2)

I also observe that there is different type of problem with v7.6.2:
Check: [CASE0069948](https://bluetooth.service-now.com/nav_to.do?uri=x_bsig_case_manage_case.do?sys_id=ba4507401b239410186a4000cd4bcb97%26sysparm_view=Self_Service_2)

sjanc, 2021-06-11:
This test pass on latest versions, closing.

closed 2021-06-11 by sjanc