auto-pts/auto-pts #394: secure connection test cases automation

jsplyy, 2020-06-23:
@michal-narajowski @carlescufi 

The PTS dongle support Bluetooth 4.2 
and I try to add secure connection cases like SM/MAS/SCJW/BV-01-C in "auto-pts\ptsprojects\zephyr\sm.py", which could pass automatically.

        ZTestCase("SM", "SM/MAS/SCJW/BV-01-C",
                  pre_conditions +
                  [TestFunc(btp.gap_set_io_cap, IOCap.display_only),
                  TestFunc(btp.gap_set_bondable_off),
                  ],
                  generic_wid_hdl=sm_wid_hdl),

So on my local, all the secure connection test cases could pass automatically with auto-pts, do you know why did not the auto-pts support sc test cases?

If you confirm it works on your side, maybe I could submit one PR for this,

MariuszSkamra, 2020-06-25:
Feel free to submit PR

KKopyscinski, 2020-10-06:
@MariuszSkamra I think this can be closed now

MariuszSkamra, 2020-10-06:
resolved

closed 2020-10-06 by MariuszSkamra