auto-pts/auto-pts #406: L2CAP/COS/CFC/BV-02-C: le_psm does not match ixit and

kromancer, 2020-09-29:
As of PTS v7.6.2 the test's implementation has a defect, reported in [CASE0069961](https://bluetooth.service-now.com/nav_to.do?uri=x_bsig_case_manage_case.do?sys_id=368202101b23d410186a4000cd4bcb09%26sysparm_view=Self_Service_2).

As the case indicates, the defect will be addressed in v7.6.3.
However, trying the proposed patch reveals 2 further issues on the upper tester:

1. The [le_psm](https://github.com/intel/auto-pts/blob/master/ptsprojects/zephyr/l2cap.py#L37) does not match the value in the IXIT. It seems like it should be set to 37. (Thinking out loud: Would it be possible to somehow parse the workspace file to extract the IXIT values instead of redefining them?)
2. The data regex for [hdl_wid_37](https://github.com/intel/auto-pts/blob/master/wid/l2cap.py#L99) seems to be wrong.
It seems like it should be something like `re.compile(r"send\sthe\sdata\s([0-9a-fA-F]+)")`

After applying the updated L2CAP.ets attached in the case + the aforementioned 2 patches, the test passes:
[updated_ets_and_fixed_upper_tester_bug_L2CAP_COS_CFC_BV_02_C_2020_09_29_11_21_46.zip](https://github.com/intel/auto-pts/files/5298168/updated_ets_and_fixed_upper_tester_bug_L2CAP_COS_CFC_BV_02_C_2020_09_29_11_21_46.zip)

michal-narajowski, 2020-11-24:
Hi @kromancer,

could you submit a PR to this repo too? Thanks!

Best,
Michał

kromancer, 2020-11-24:
Michal I think we fixed this one here:
https://github.com/kromancer/auto-pts/commit/b074dc710f73ce4b64646c946977b3ae556a621e

michal-narajowski, 2020-11-25:
Ok, great! Then I'm closing this issue.

closed 2020-11-25 by michal-narajowski