auto-pts/auto-pts #410: [RFC] Refactor auto-pts

michal-narajowski, 2020-10-06:
The project could use a large refactor. I have a few ideas:
- **Get rid code duplication** - We don't need to list every test case that is supported. A PTS workspace contains all supported test cases. By default we should have some generic wid handler that is common to all platforms and maybe some preconditions (some requirements for test case are not communicated through wids). If a platform needs some specific behavior then it should be implemented but 90% of the time it would be just default handlers and this removes a lot of code. After this change adding support for new platforms would be much easier (to implement, to test and to review).
- **Get rid of globals** - This makes the dependencies complicated and adding new features is problematic. The design is not clean. As an example: iutctl. 
- **Get rid of ZTestCase** - It should be made more generic.
- **Split btp.py into multiple files** - This file is huge and indexing it takes a lot of time.
- **Merging bot client and regular client** - We could have a single client that has all the features of a bot client and the ease of use of the regular one. We could move platform specific stuff to iutctl or some other module that is implemented by each platform. We could add command line options to bot that will allow to override some settings that are specifiec in configuration file. So the default would be a config file but if we want we could specify CLI options and it will behave just like the regular client.

mringwal, 2020-10-06:
That's a great idea. While I have no / not much experience with the other points, I've seen and wondered about the per-project code for setting up the test cases and the ZTestCase. 

If there's a generic profile.py and profile_wid.py, tweaks could be implemented per-project (if needed). I have cases with BTstack where I've disabled a test as it causes auto-pts to hang and I'd like to keep it active in the PTS workspace to get back to fix that one later, so, a way to disable a test case (aka 'blocklist') instead of individually enabling a case would be useful, too.

michal-narajowski, 2020-11-24:
Hi @mringwal, 

thanks for the suggestion! I will add it to a list of possible improvements we can make.

Best,
Michał