auto-pts/auto-pts #417: When I was using autoPTS, the server encountered a problem: PTSCONTROL_E_PROJECT_NOT_FOUND exception occurred when sending PIXIT parameters. Can you look at this problem?

gaoheyang, 2020-11-13:
When I was using autoPTS, the server encountered a problem: PTSCONTROL_E_PROJECT_NOT_FOUND exception occurred when sending PIXIT parameters. Can you look at this problem? The following is the server log information

[autoptsserver.log](https://github.com/intel/auto-pts/files/5534152/autoptsserver.log)

michal-narajowski, 2020-11-24:
Hi @gaoheyang, 

sorry for the delay. Looks like auto-pts is trying to set PIXIT parameters for all projects and it fails on SM, so probably you don't have SM project added in your workspace. Let me know if that's the case. 

If that's the issue then it should be fixed so that we don't initialize projects that are not supported.

Best,
Michał Narajowski

sjanc, 2022-08-23:
stale, closing

please reopen if this is stil an issue

closed 2022-08-23 by sjanc