auto-pts/auto-pts #427: [SM/CEN/PKE/BV-01-C] PTS INCONC - There is empty Passkey Code

ryanjh, 2021-01-26:
PTS version: 8.0.0.29

Test Case:
SM/CEN/PKE/BV-01-C
SM/MAS/PKE/BI-01-C

Error:
INCONC

Description:
Verify that the IUT performs the Passkey Entry pairing procedure correctly as initiator. During the phase 2 passkey pairing process, Lower Tester displays the 6-digit passkey while the IUT prompts user to enter the 6-digit passkey. Sometimes, it failed on the responder side about "There is empty Passkey Code".

Log:
autoptsserver.log
autoptsclient_bot.log

ryanjh, 2021-03-10:
any comment about this issue?

michal-narajowski, 2021-03-10:
Hi @ryanjh, I reproduced this issue but honestly I don't see a difference in automation behavior between PASS and INCONC run.

ryanjh, 2021-03-10:
hi @michal-narajowski, could it be related to a timing issue when using automation framework? This test passed consistently on my side when it is verified by running Profile Tuning Suite (PTS) step by step. Do you see the same behavior as well?

michal-narajowski, 2021-03-11:
@ryanjh This issue looks similar to something I encountered before in SM: 
`CASE0060005 CASE0061902`. If you delay the response to that MMI the PTS failed the test case anyway (without waiting for automation's response.

ryanjh, 2021-03-11:
thanks for sharing those cases. The issue of CASE0060005 can't be reproduced by v7.5.1. CASE0061902 is used to fix the "No IO bug". I also created a case for SM/MAS/SCOB/BV-01-C (CASE0070257).

When using Profile Tuning Suite (PTS) to verify SM/CEN/PKE/BV-01-C, I tried to input passkey with delays on purpose. In this case, PTS was working fine. Based on their feedback, they pretty sure is relates how those mmi is handle in the automation.

ryanjh, 2022-08-23:
Close this issue as it has been fixed

closed 2022-08-23 by ryanjh