auto-pts/auto-pts #511: Implement new tests for Zephyr (PTS v8.1)

KKopyscinski, 2021-07-14:
New tests are:

- GAP/DISC/LIMM/BV-02-C
- GAP/CONN/CPUP/BV-10-C
- GAP/SEC/SEM/BV-45-C
- GAP/SEC/AUT/BV-25-C
- GAP/SEC/AUT/BV-26-C
- GAP/SEC/AUT/BV-27-C
- GAP/SEC/AUT/BV-28-C
- GAP/PRIV/CONN/BV-12-C
- GAP/ADV/BV-18-C
- GATT/CL/GAI/BI-01-C
- GATT/CL/GPM/BV-11-C
- GATT/SR/GAR/BI-45-C
- L2CAP/LE/CFC/BV-22-C
- L2CAP/LE/CFC/BV-23-C
- L2CAP/LE/CFC/BV-24-C
- L2CAP/LE/CFC/BV-25-C
- L2CAP/LE/CFC/BV-26-C
- L2CAP/LE/CFC/BV-27-C
- L2CAP/LE/CFC/BV-28-C
- L2CAP/LE/CFC/BV-29-C
- L2CAP/LE/CFC/BV-30-C
- L2CAP/LE/CFC/BV-31-C
- L2CAP/LE/CFC/BI-02-C
- L2CAP/LE/CID/BI-01-C
- L2CAP/ECFC/BV-28-C
- L2CAP/ECFC/BV-29-C
- L2CAP/ECFC/BV-30-C
- L2CAP/ECFC/BV-31-C
- L2CAP/ECFC/BV-32-C
- L2CAP/ECFC/BV-33-C
- L2CAP/ECFC/BV-34-C
- L2CAP/ECFC/BV-35-C
- L2CAP/ECFC/BV-38-C
- L2CAP/ECFC/BV-41-C
- L2CAP/ECFC/BI-07-C
- L2CAP/TIM/BV-03-C
- MESH/NODE/CFG/MP/BV-03-C
- SM/CEN/KDU/BI-02-C
- SM/CEN/KDU/BI-03-C
- SM/PER/KDU/BI-02-C
- SM/PER/KDU/BI-03-C

michal-narajowski, 2021-08-03:
What's the current state of this issue?

michal-narajowski, 2022-02-03:
Closing as all the tests for 8.1 were implemented and we are currently updating to 8.2.

closed 2022-02-03 by michal-narajowski