auto-pts/auto-pts #544: GAP/CONN/NCON/BV-02-C Failed

LingaoM, 2021-08-17:
[autopts-iutctl-zephyr.log](https://github.com/intel/auto-pts/files/6999047/autopts-iutctl-zephyr.log)
[GAP_CONN_NCON_BV-02-C_2021_08_16_02_02_41.log](https://github.com/intel/auto-pts/files/6999049/GAP_CONN_NCON_BV-02-C_2021_08_16_02_02_41.log)
[GAP_CONN_NCON_BV_02_C_2021_08_16_17_03_09.zip](https://github.com/intel/auto-pts/files/6999067/GAP_CONN_NCON_BV_02_C_2021_08_16_17_03_09.zip)

mkasenberg, 2021-08-20:
I can reproduce this fail when missing config overlay with those:
CONFIG_BT_PRIVACY=y
CONFIG_BT_RPA_TIMEOUT=30

LingaoM, 2021-08-20:
so, we need add this config to dot/zephyr.conf file?

mkasenberg, 2021-08-20:
I see it is already there in recent version of bot/config.py.zephyr.sample at:
"privacy.conf": {
        "overlay": {
            'CONFIG_BT_PRIVACY': 'y',
             'CONFIG_BT_RPA_TIMEOUT': '30',
         },
        "test_cases": [
            'GAP/PRIV/CONN/BV-10-C',
            'GAP/PRIV/CONN/BV-11-C',
            'GAP/CONN/ACEP/BV-03-C', # As of PTS v7.6.2, not supported
            'GAP/CONN/DCEP/BV-05-C', # As of PTS v7.6.2, not supported
            'GAP/CONN/GCEP/BV-05-C', # As of PTS v7.6.2, not supported
            'GAP/CONN/NCON/BV-02-C',
            'GAP/CONN/UCON/BV-06-C',
            'GAP/BROB/BCST/BV-03-C',
            'GAP/BROB/BCST/BV-04-C',
            'GAP/BROB/OBSV/BV-06-C',
            'GAP/DISC/RPA/BV-01-C',
            'SM/CEN/KDU/BV-05-C',
            'SM/CEN/KDU/BV-10-C',
            'SM/CEN/KDU/BV-11-C',
            'SM/PER/KDU/BV-02-C',
            'SM/PER/KDU/BV-08-C',
        ]
    },

sjanc, 2021-09-10:
for the record,  we should have tests configuration separated from bot config so that people always run with latest tests settings 

https://github.com/intel/auto-pts/issues/523
https://github.com/intel/auto-pts/pull/557

So hopefully this kind of errors will be reduced once #557 is merged:)

sjanc, 2021-09-16:
is it fixed now? can this issue be closed?

closed 2021-09-16 by LingaoM