auto-pts/auto-pts #740: Extend BTP specification with Periodic Advertising

KKopyscinski, 2022-04-20:
This is required by GAP/PADV tests

jsplyy, 2023-01-17:
Hi @KKopyscinski ,  

Will you plan to fix this issue recently?
and how to test periodic adv feature before this issue is resolved?
Is there any rules/guideline to define btp protocol, if we want to add periodic feature support.

KKopyscinski, 2023-01-17:
Hi @jsplyy 

I started working on periodic adv for Mynewt at some point and created PR for this https://github.com/intel/auto-pts/pull/785. However, I got no more time to work on it, so I guess it's just left hanging like that. AFAIK this PR was working but needed some improvements. Of course you'd need tester application to flash board with, that supports new BTP commands, like this: https://github.com/apache/mynewt-nimble/pull/1311.

This issue was created as reminder that newly added commands will need to be documented at some point, once it's established what do we need them to do (what arguments will they take, what responses we'd like to expect etc.)

I think there are no specific guidelines to define BTP protocol, this protocol is created here, as needed. Of course you want to follow conventions established by already added commands, but I think that's all.

jsplyy, 2023-02-28:
Thanks @KKopyscinski .
Hi @sjanc , Do you know if any one have working on updating this pr for zephyr PADV pts testing?
If no, I could try to update tester app and script to make it work.

closed 2023-12-06 by sjanc