auto-pts/auto-pts #750: GAP/BOND/BON/BV-03-C pairing fail , with add "TestFunc(sleep, 1, start_wid=118)" it PASS

twiceand, 2022-05-18:
**summary:** 
I tested cases with nRF52840 board ,( I will I porting auto-pts to test nxp rt1060 board later)
**Reproduce**
server: python .\autoptsserver.py
client: python ./autoptsclient-zephyr.py zephyr-master -t COM34 -b nrf52 -c GAP/BOND/BON/BV-03-C
**analysis**
- it can not distribute the LTK with out adding  "TestFunc(sleep, 1, start_wid=118)" in autopts/ptsproject/zephyr/gap.py after line 
>ZTestCase("GAP", "GAP/BOND/BON/BV-03-C",
cmds=pre_conditions + init_gatt_db +
[TestFunc(btp.gap_set_io_cap, IOCap.display_only)],
generic_wid_hdl=gap_wid_hdl),

>ZTestCase("GAP", "GAP/BOND/BON/BV-03-C",
cmds=pre_conditions +
[TestFunc(btp.gap_set_io_cap, IOCap.no_input_output),
# give some time before returning wid 118 to allow pair to finish
TestFunc(sleep, 1, start_wid=118)],
generic_wid_hdl=gap_wid_hdl),

michal-narajowski, 2022-05-26:
What is the description of WID 108?

sjanc, 2023-03-21:
is this stil an issue? that test seems to be quite stable now

sjanc, 2023-04-13:
please reopen if this is stil an issue for you

closed 2023-04-13 by sjanc