auto-pts/auto-pts #754: Fix overlay application on test if more than one overlay applies to same test

KKopyscinski, 2022-05-30:
There is a case where one overlay contains group of tests and other it's subgroup. Then tests get overlay of "master" group and subgroup is not build. We should add solution that prioritizes longer test prefixes in applying overlay.

Example:
```python
overlay1.conf: {
    "overlay": {
        'CONFIG_VAL': 'x',
    },
    "test_cases" : [
        'XXXX/YY/ZZ'
    ]
}

overlay2.conf: {
    "overlay": {
        'CONFIG_VAL': 'y',
    },
    "test_cases" : [
        'XXXX/YY/ZZ/BV-01-C'
    ]
}
```
All `XXXX/YY/ZZ` tests should be built with `overlay1`, but `XXXX/YY/ZZ/BV-01-C` should be build with `overlay2`.

sjanc, 2022-05-30:
my gut feeling is that overlays should be always explicit when it comes to test names,  ie it should always lists every single test is affects

KKopyscinski, 2022-05-30:
In current state it seems like we must do this, but maybe it's worth to add some flexibility as test groups? It would future proof overlay, as more test cases may be added later, and probably they will use same overlay as others