auto-pts/auto-pts #807: Review errata.yaml

KKopyscinski, 2022-09-26:
Some tests are passing and erratas are resolved. List of issues should be reviewed and updated.

twiceand, 2022-09-29:
does errata case can't test with the auto-pts script?
does zephyr has another way to test with errata cases ?

michal-narajowski, 2022-09-29:
@twiceand Currently there is no way to automatically update ETS files that fix PTS issues if this is what you are asking. If a test is failing because of a PTS issue then it has to be manually tested with a patch provided by SIG. If it is a TS errata or spec errata then a waiver needs to be issued for that test. Not sure if this answers your questions.

twiceand, 2022-09-29:
I have another question, I am trying to add cases recently, how should I handle the corresponding wid requests, is there any reference document?

michal-narajowski, 2022-09-29:
@twiceand Good question. We should add a contribution guide for new developers.

If you want to contribute cases to the public repo and to a public project like zephyr:
- Add new generic WID handlers to `<root>/autopts/wid/<profile>.py`.
- If a test case requires some specific preconditions or some setup, add an entry to `custom_test_cases` list in `<root>/autopts/ptsprojects/<project>/<profile>.py`.

Basically, mimic what is already done.
If you have more questions feel free to reach out.

twiceand, 2022-09-29:
> @twiceand Good question. We should add a contribution guide for new developers.
>
> If you want to contribute cases to the public repo and to a public project like zephyr:
>
> * Add new generic WID [...]
Yes, I am using the auto-pts script to verify the integrity of the edgefast-based NXP Bluetooth stack
We are very interested in this and would like to support the profile part and the classic BT part as well
But when we tried to add the case ourselves, we didn't find the correspondence between the wid and the case
And we want to contribute and maintain the nxp branch base on auto-pts

twiceand, 2022-09-29:
Yes, I am using the auto-pts script to verify the integrity of the edgefast-based NXP Bluetooth stack
We are very interested in this and would like to support the profile part and the classic BT part as well
But when we tried to add the case ourselves, we didn't find the correspondence between the wid and the case
And we want to contribute and maintain the nxp branch base on auto-pts

justin-nxp, 2022-09-29:
Yes, I am using the auto-pts script to verify the integrity of the edgefast-based NXP Bluetooth stack
We are very interested in this and would like to support the profile part and the classic BT part as well
But when we tried to add the case ourselves, we didn't find the correspondence between the wid and the case
And we want to contribute and maintain the nxp branch base on auto-pts

michal-narajowski, 2022-09-29:
@twiceand Ok, so basically you will need to add a ptsproject, similar to `<root>/autopts/ptsprojects/zephyr/`. 
What you will need:
- A tester application implemented on your RTOS running on your board of choice. This application needs to implement BTP protocol to communicate with auto-pts, see `<root>/doc/btp_spec.txt`.
- A PTS workspace file, see `<root>/autopts/workspaces/`
- A way to set up communication between auto-pts and your board. Most of our projects use UART-based communication (as an example look at `<root>/autopts/ptsprojects/zephyr/iutctl.py`.
- Then you need to create a list of test cases per each profile, see `<root>/autopts/ptsprojects/zephyr/gap.py`.
- And you will also need a starting point for the client application, see `<root>/autoptsclient-zephyr.py`. (this is a simple command line tester, once you have everything set up you can use a bot to run a whole suite of tests and schedule weekly test runs for example).

The correspondence between wid and the test case is defined in `<root>/autopts/ptsprojects/<project>/<profile>.py`. There you create an `TestCase` object and you specify a `generic_wid_hdl`. By default you can use the once implemented in `<root>/autopts/wid/`.

If you have any work-in-progress branch that you would like to share then we might be able to give you more specific advice on how to move on.

justin-nxp, 2022-09-29:
We are in the process of tidying up the local branch and will launch pull-request integration into the existing auto-pts when we are done

I know from other places that there are weekly auto-pts test results about zephyr, I want to get the weekly test results, how should I add my email to the list alert list?

michal-narajowski, 2022-09-29:
Do you want access to weekly test results of zephyr or do you want to do weekly runs of your test suite and an email summary of that?

justin-nxp, 2022-09-29:
I would like to access zephyr's weekly test results, I don't know where to get them, if you know please do let me know

michal-narajowski, 2022-09-29:
@justin-nxp Unfortunately this mailing list is not public.

carlescufi, 2022-09-29:
@justin-nxp Since you are an NXP employee you do have access. Please contact @dleach2 so he can make a request to me in order to add you to the mailing list.

justin-nxp, 2022-09-30:
I have tried to contact @dleach02 through the intranet, thank you for your help!

KKopyscinski, 2022-10-03:
Solved by https://github.com/intel/auto-pts/issues/807

closed 2022-10-03 by KKopyscinski