auto-pts/auto-pts #816: migrate erartas to bluetooth.atlassian.net links

sjanc, 2022-10-11:
BTSIG enabled new errata system and old one is read-only now,  we should update errata.yaml with links to new system

justin-nxp, 2022-10-15:
I've been working on errta-related issues recently, and I found that the errta links https://www.bluetooth.org/tse/errata_view.cfm?errata_id=18349

for these cases
GAP/SEC/SEM/BV-62-C: TSE18349
GAP/SEC/SEM/BV-63-C: TSE18349
GAP/SEC/SEM/BV-64-C: TSE18349
GAP/SEC/SEM/BV-65-C: TSE18349
GAP/SEC/SEM/BV-66-C: TSE18349
GAP/SEC/SEM/BV-67-C: TSE18349

What does status  "incoporate" mean and what does it stand for? 
Do these errata cases mean that they are exempt when submitting test evidence?
[image]

justin-nxp, 2022-10-17:
And what does the "CASE" means, Can i have a link of it 
GAP/SEC/AUT/BV-25-C: CASE0072001

michal-narajowski, 2022-10-17:
"CASE" numbers are PTS issue numbers in the old PTS support portal.

justin-nxp, 2022-10-17:
> "CASE" numbers are PTS issue numbers in the old PTS support portal.

Yes，The new system seems to only have the ability to view the issues that you have submitted, there is no way to view the previous issues, but I would like to view some details about CASE0072001, how should I do it who can tell me？

michal-narajowski, 2022-10-17:
@KKopyscinski @sjanc Can you check if this is your errata?

michal-narajowski, 2022-10-17:
@KKopyscinski Maybe you have it in the new portal? Or did they not migrate old issues? Anyway, I agree with @KKopyscinski that it can be removed as fixed.

KKopyscinski, 2022-10-17:
I don't have any cases for GAP/SEC/AUT/BV-25-C, I checked new portal first, then searched email. I have other submissions from old system in email, so I guess I haven't filed this one. I don't think they migrated anything, I guess old cases are lost

justin-nxp, 2022-10-18:
errata system can be login in thanks
[image]

sjanc, 2022-10-19:
FYI I've removed screenshots from internal Bluetooth SIG systems as those are not public resources.

As for issues in errata yaml:
SE stands for (Bluetooth) Specification Errata
TSE stands for Test Set Errata (fixes in test specification)
CASE is legacy PTS issue (bugs in PTS), this is no longer accessible but all of those should be fixed already
Request ID is "new" PTS issue reporting system, unfortunately it is not possible to make such issue accessible to other BT SIG members (due to way how BT SIG support system works)

Only SE and TSE are using new jira-based system (at least for now).

As for how errata integration process work I suggest you refer to Bluetooth SIG material and their support system.

justin-nxp, 2022-10-19:
@sjanc According to your description, are these PTS CASE issues automatically repaired with the update of the PTS version without follow-up? Will the unresolved parts of the old PTS issue be migrated to the new issue system?

sjanc, 2022-10-19:
> @sjanc According to your description, are these PTS CASE issues automatically repaired with the update of the PTS version without follow-u

I think all PTS issues reported in legacy system (CASEXXX) were fixed and should be removed from errata yaml,  ie affected tests are passing with latest PTS.

> Will the unresolved parts of the old PTS issue be migrated to the new issue system?

Don't know. You should ask PTS folks or BT SIG about this

closed 2023-12-01 by sjanc