auto-pts/auto-pts #841: GATT/CL/GPM/BV-11-C has not supported

justin-nxp, 2022-12-15:
Hello developers,this was an new caes added GATT/CL/GPM/BV-11-C,This is a new case added recently, the script is not yet supported and involves hdl_wid_165 hdl_wid_153. Will these hdl be supported subsequently?

sjanc, 2022-12-15:
since those are using concurrent gatt read we first need to move both autopts and zephyr tester app to new BTP GATT client service

see:
https://github.com/intel/auto-pts/issues/582
https://github.com/intel/auto-pts/blob/master/doc/btp_gatt_client.txt

MariuszSkamra, 2023-01-03:
@sjanc Would it require much work to add bare minimum support for new BTP GATT Client API in Zephyr just to pass this 1 particular test case?

michal-narajowski, 2023-01-03:
@sjanc It might be possible to just implement whatever command is needed to pass this test (e.g. GATT Read from the new BTP service).

sjanc, 2023-01-03:
yes, in short term, although in general IUT should support only new services or only old service
and in long term all tests should use new services, and only fallback to old if IUT doesn't support it (as long as maintaining this isn't huge burden)

jsplyy, 2023-01-04:
> since those are using concurrent gatt read we first need to move both autopts and zephyr tester app to new BTP GATT client service
> 
> see: #582 https://github.com/intel/auto-pts/blob/master/doc/btp_gatt_client.txt

hi @sjanc what is the difference in this new btp_gatt_client.txt?
I could not find the channel option for gatt api/command, or bearer params.

how to use the concurrent gatt read with zephyr bluetooth api?     
I have tried to understand this pr, https://github.com/zephyrproject-rtos/zephyr/pull/45332 
the zephyr bluetooth api, seem only to set channel as any or UNENHANCED_ONLY or ENHANCED_ONLY, other than set one special channel/bearer id?

hermabe, 2023-01-04:
> > since those are using concurrent gatt read we first need to move both autopts and zephyr tester app to new BTP GATT client service
> > see: #582 https://github.com/intel/auto-pts/blob/master/doc/btp_gatt_client.txt
> 
> hi @sjanc what is the difference in this new btp_gatt_client.txt? I could not find the channel option for gatt api/command, or bearer params.
> 
> how to use the concurrent gatt read with zephyr bluetooth api? I have tried to understand this pr, [zephyrproject-rtos/zephyr#45332](https://github.com/zephyrproject-rtos/zephyr/pull/45332) the zephyr bluetooth api, seem only to set channel as any or UNENHANCED_ONLY or ENHANCED_ONLY, other than set one special channel/bearer id?

There is no api for choosing a specific channel. We did not see the use case for that and the enhanced/unenhanced switch is sufficient for both the audio use cases and for the PTS tests.

The BTP protocol does not currently have any way to specify the bearer to use, but that could be added. See [this comment](https://github.com/intel/auto-pts/pull/620#issuecomment-1049610538) and responses.

----

GATT/CL/GPM/BV-11-C requiresGATT caching on the client side. This is not implemented in the Zephyr host stack and will have to be done in the application. I am not sure it makes sense to run this qualification test for something the host does not support.

See enhancement issues in zephyr relating to client side GATT caching: https://github.com/zephyrproject-rtos/zephyr/issues/31306, https://github.com/zephyrproject-rtos/zephyr/issues/46695, https://github.com/zephyrproject-rtos/zephyr/issues/44587

sjanc, 2023-01-18:
if I recall correctly ICS table defines only single option for supporting "database hash" and both GATT client and server scenarios depend on it. So disabling client side would require disabling also server side (which is already supported and we want to validate it)

we don't plan to add support for GATT client caching in tester app as this feature is no supported by zephyr host yet

but TBH I think we should create test set errata so that client and server support are independent