auto-pts/auto-pts #846: where to find the ERRATA Request ID 16410 for the L2CAP/ECFC/BV-32-C failure in CI report

jsplyy, 2023-01-05:
we could see L2CAP/ECFC/BV-32-C result with comment "FAIL - ERRATA Request ID 16410" in CI email.
where to find the ERRATA Request ID 16410? the Request ID is not TSE ID, right?

https://www.bluetooth.org/tse/errata_view.cfm?errata_id=16410
this seems to be another issue, for GAP/SEC/AUT/BV-20-C.

sjanc, 2023-01-05:
No, this is PTS issue tracker.
Unfortunately new PTS issue tracking system doesn't allow for public requests... (this was reported to PTS but they didn't give any ETA on when such feature could be added)

jsplyy, 2023-01-06:
hi @sjanc , I have access to the two bt sig errata system, 
https://www.bluetooth.org/tse/errata_search.cfm
https://bluetooth.atlassian.net/jira/software/c/projects/ES/issues

the new PTS issue tracking system is one of them ?   or another ticket management system?

sjanc, 2023-01-06:
BT SIG errata system and PTS issue tracking are separate systems

closed 2023-02-06 by sjanc