auto-pts/auto-pts #852: check_args_btpclient_path needed for --btpclientpath argument

mringwal, 2023-01-31:
E.g. in `autopts/ptsprojects/bluez/iutclt.py` sets

`CLI_SUPPORT = ['btp_py', 'btpclient_path']`

which causes `cliparser.py` to call `check_args_btpclient_path()`, if --btpclient_path is used, but which doesn't exist and gives an error.

Interestingly, 'btp_py' in CLI_SUPPORT does not allow for additional arguments, but there's a `check_args_btp_py()` function. Could it be that the `btp_py` is a left-over, and it's enough to rename check_args_btp_py into check_args_btpclient_path? While we're looking at this, I guess 'btp_py' could then also be removed from bluez/iutctl.py.

mkasenberg, 2023-01-31:
Yes, this is a leftover. Back in time btp_py was the name for btpclient_path. The name was confusing, because the btp script of course could be other than .py file.

mringwal, 2023-01-31:
Would you like me to create a RP for this, or do you want to fix it?

mkasenberg, 2023-01-31:
I don't have a Bluez environment set up for testing, so I'd appreciate your help, thanks!

mringwal, 2023-01-31:
Sure. Tested with our btpclient, which is based on the bluez implementation.

PR #853 (2023-01-31): cliparser: fix btpclient_path support
Drops unused 'btp_py' flag in bluez/iutclt.py.

Fixes #852

Tested with btstack/iuctl.py which is derived from bluez/iuctl.py.
closed 2023-01-31 by mringwal