auto-pts/auto-pts #912: Read-only file system

eswin123, 2023-05-06:
2023/05/06 14:44:57 socat[3740] E open("/dev/ttyS90", 01002, 0666): Read-only file system

eswin123, 2023-05-06:
then BTP TIMEOUT

mkasenberg, 2023-05-07:
Does your user belong to the "dialout" group (which has permission to access the serial ports)?

eswin123, 2023-05-07:
> Does your user belong to the "dialout" group (which has permission to access the serial ports)?

hi，i do the test in the window environment，just have administrator and uesr group（have no dialout group），and the “autopts”  require user rights when it running。

mkasenberg, 2023-05-08:
Oh, I see. The `/dev/ttyS90` maps to `COM91`. Double check if the `COM91` represents your board. If you are sure it is right, lets check if your socat.exe properly works. 

Check your socat.exe version:
`socat.exe -V`
My socat.exe version is 1.7.3.2, downloaded from https://sourceforge.net/projects/unix-utils/files/socat/1.7.3.2/ and it's working fine so far, under Windows 10 or 11.

Next, in a first console create a server:
`socat.exe - TCP-LISTEN:65123`

In a second console run:
`netstat -a | grep 65123`
The output should be like this:
```  TCP    0.0.0.0:65123          DESKTOP-KV46KLN:0      LISTENING```

In a third console run:
`mode COM91 BAUD=115200 PARITY=n DATA=8 STOP=1`
because socat.exe has issues with changing the baud rate. Then run
`socat.exe -x -v tcp:0.0.0.0:65123 /dev/ttyS90,raw,b115200`
to connect to the server. There will be no output yet. Now restart board with `nrfjprog.exe -r` or the reset button.
If the board is flashed with the tester app, then in the output of the third console should appear:
```
< 2023/05/08 09:01:35.433286  length=5 from=0 to=4
 00 80 ff 00 00                                   .....
--
```
That means the sockat.exe works as expected.

But the "Read-only file system" in the error suggests its issue with the COM. Sometimes I mistakenly select a wrong one myself. Then I receive similar messages like:
`2023/05/08 09:04:39 socat[4300] E tcgetattr(6, 0xffffc5f0): Invalid argument`
or
`2023/05/08 09:07:50 socat[13156] E open("/dev/ttyS3", 012, 0666): Permission denied`

sjanc, 2023-11-29:
reopen if this is stil an issue

closed 2023-11-29 by sjanc