auto-pts/auto-pts #1021: Exceptions while running client

omkar3141, 2023-10-05:
Hi,

I frequently see following exceptions while running auto-pts client on windows.

```
$ python ./autoptsclient-zephyr.py zephyr-master-nordic --rtt-log -t COM4 -b nrf52 -c MMDL/SR/LLC/BV-12-C -S 65000 65002 -C 65001 65003 
(2846176583344) Starting PTS 127.0.0.1:65000 ...
(2846176642528) Starting PTS 127.0.0.1:65002 ...
(2846176583344) OK
Traceback (most recent call last):
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 1175, in start
    return self.main(args)
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 1205, in main
    init_pts(self.args, self.ptses, tc_db_table_name)
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 501, in init_pts
    raise Exception(exeption_msg)
Exception: unknown encoding: idna

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./autoptsclient-zephyr.py", line 39, in <module>
    main()
  File "./autoptsclient-zephyr.py", line 35, in main
    client.start()
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 1180, in start
    shutdown_pts(self.ptses)
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 511, in shutdown_pts
    proxy.unregister_xmlrpc_ptscallback()
  File "xmlrpc\client.py", line 1109, in __call__
  File "xmlrpc\client.py", line 1450, in __request
  File "xmlrpc\client.py", line 1153, in request
  File "xmlrpc\client.py", line 1169, in single_request
  File "xmlrpc\client.py", line 1341, in parse_response
  File "xmlrpc\client.py", line 655, in close
xmlrpc.client.Fault: <Fault 1: "<class 'AttributeError'>:'NoneType' object has no attribute 'unset_callback'">
```

Could you please address this issue?

mkasenberg, 2023-10-05:
This is not a typical exception, looks more related to Windows version or console encoding. Could you check what versions do you use by running `chcp` and `ver` in CMD? You will got something like this:
```
C:\Users\Codecoup>chcp
Active code page: 852
C:\Users\Codecoup>ver
Microsoft Windows [Version 10.0.19045.3208]
```

You can look up the autoptsclient-zephyr.log and autopstserver_6500_65002.log files to find the exact line that caused the problem too, there should be the original exception.

mkasenberg, 2023-10-05:
Or maybe it is a python issue, what version do you use? Did the command `pip3 install -r ./autoptsclient_requirements.txt` throw any exception?

omkar3141, 2023-10-05:
> This is not a typical exception `Exception: unknown encoding: idna`, looks more related to Windows version or console encoding. Could you check what versions do you use by running `chcp` and `ver` in CMD? You will got something like this:
> 
> ```
> C:\Users\Codecoup>chcp
> Active code page: 852
> C:\Users\Codecoup>ver
> Microsoft Windows [Version 10.0.19045.3208]
> ```
> 
> You can look up the autoptsclient-zephyr.log and autopstserver_6500_65002.log files to find the exact line that caused the problem too, there should be the original exception.

I'm running auto PTS inside Git-bash, may be that is the problem?

omkar3141, 2023-10-05:
> pip3 install -r ./autoptsclient_requirements.txt

No exception. It works fine in Git bash, all requirements are already installed.

omkar3141, 2023-10-05:
This exception occurs at random.
```
Exception: unknown encoding: idna
```

Sometimes client starts without any issues, and most of the times it throws this exception.

mkasenberg, 2023-10-05:
@sjanc found similar issue here https://github.com/psf/requests/issues/3578. Could you try to add `import encodings.idna` at the begging of the `autoptsclient-zephyr.py` script?

sjanc, 2023-12-06:
is this stil an issue?

omkar3141, 2023-12-07:
Yes. I always hack it off using suggestion mentioned in "https://github.com/auto-pts/auto-pts/issues/1021#issuecomment-1748722897" and then issue never appears.

PR #1075 (2023-12-07): Fix Exception: unknown encoding: idna
Workaround for Exception: unknown encoding: idna that appears randomly in embedded Python distribution.
Fixes #1021.
commit d16c9733 (2023-12-07): Fix Exception: unknown encoding: idna

Workaround for Exception: unknown encoding: idna that appears
randomly in embedded Python distribution.
Fixes #1021.
closed 2023-12-07 by sjanc