auto-pts/auto-pts #1023: Frequent errors "IUT ready event NOT received!"

omkar3141, 2023-10-05:
Hi,

I'm using auto-pts on windows, and when the client starts, I get following error every time and I'm unable to run any tests.

```
2023-10-05 13:09:02,528 LT1-thread root DEBUG __init__.py               48    reset                     : About to reset DUT: 'nrfjprog -r -s 683326223'
2023-10-05 13:09:03,240 LT1-thread root DEBUG testcase.py               246   start                     : Starting test function: class=<class 'autopts.ptsprojects.testcase.TestFunc'>, func=<bound method ZephyrCtl.wait_iut_ready_event of <autopts.ptsprojects.zephyr.iutctl.ZephyrCtl object at 0x0000019B7F011850>> start_wid=None stop_wid=None post_wid=None skip_call=None call_count=1 args=(False,) kwds={}
2023-10-05 13:09:03,240 LT1-thread root DEBUG testcase.py               258   start                     : Test function parameters: args=(False,), kwds={}
2023-10-05 13:09:33,258 LT1-thread root DEBUG iutctl.py                 247   stop                      : <class 'autopts.ptsprojects.zephyr.iutctl.ZephyrCtl'>.stop
2023-10-05 13:09:33,258 LT1-thread root DEBUG __init__.py               48    reset                     : About to reset DUT: 'nrfjprog -r -s 683326223'
2023-10-05 13:10:04,004 LT1-thread root DEBUG iutctl.py                 265   stop                      : IUT ready event NOT received!
2023-10-05 13:10:04,480 LT1-thread root ERROR client.py                 901   run_test_case_thread_entry : IUT ready event NOT received!
Traceback (most recent call last):
  File "E:\NRD_repos\auto-pts\autopts\client.py", line 883, in run_test_case_thread_entry
    test_case.pre_run()
  File "E:\NRD_repos\auto-pts\autopts\ptsprojects\testcase.py", line 751, in pre_run
    cmd.start()
  File "E:\NRD_repos\auto-pts\autopts\ptsprojects\testcase.py", line 260, in start
    self.func(*args, **self.kwds)
  File "E:\NRD_repos\auto-pts\autopts\ptsprojects\zephyr\iutctl.py", line 231, in wait_iut_ready_event
    raise Exception('IUT ready event NOT received!')
Exception: IUT ready event NOT received!

```

omkar3141, 2023-10-05:
[autoptsclient-zephyr_65001_65003.log](https://github.com/auto-pts/auto-pts/files/12817672/autoptsclient-zephyr_65001_65003.log)

omkar3141, 2023-10-05:
Update: We figured out that this issue occurred consistently on my DK PCA10056 with revision 2.0.1. But the issue did not happen on DK with revision 3.0.0.

It could be nice if the test firmware at least prints some initialization strings in the COM or RTT log, so that we can debug such issues whenever they occur.

sjanc, 2023-10-10:
There is debug message from tester that is printed at the very beginning  of initialization:

*** Booting Zephyr OS build v3.5.0-rc2-45-g474aa963ff52 ***
[00:00:00.251,312] <dbg> bttester: tester_init: Initializing tester
[00:00:01.671,813] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:01.671,875] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:01.671,905] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.5 Build 0
[00:00:01.672,912] <inf> bt_hci_core: Identity: D3:8E:1F:16:3C:9C (random)
[00:00:01.672,943] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:01.672,973] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff

closed 2023-11-29 by sjanc