auto-pts/auto-pts #1114: Test fails with XML-RPC ERROR

rlorenzoleal, 2024-02-13:
We're trying to run auto-pts against BlueZ  5.65 running on Ubuntu 22.04.

We're using a different PTS workspace since we don't need to run the same amount of tests as the default BlueZ workspace. 

We installed and configured BlueZ as explained [here](https://github.com/auto-pts/auto-pts/blob/master/autopts/ptsprojects/bluez/README.md). 

To start we ran all the GAP tests we have. All of them failed with `XML-RPC ERROR`. 

Doing some research I found that that error indicates an issue in communication between auto-pts client and server, however I can see both client and server responding to each other. 

What can be causing this issue? Where can I look for logs regarding server-client communication?

rlorenzoleal, 2024-02-14:
Brief update: 

I found the logs and the issue seems to be a missing PIXIT parameter, specifically: 
```
xmlrpc.client.Fault: <Fault 1: "<class 'autopts.ptscontrol.BadPixitException'>:GAP TSPX_using_private_device_address value=TRUE :\nPTSCONTROL_E_PIXIT_PARAM_NOT_FOUND">
```

Is there any documentation on how to add all the necessary PIXIT to the workspace?

sjanc, 2024-02-14:
BlueZ support is bit stale unfortunately,   I guess you would have to start with updating workspace to latest PTS

there are tutorials at [1] which may help a bit

[1] https://github.com/auto-pts/auto-pts/tree/master/doc/tutorials

iceaway, 2024-03-06:
@rlorenzoleal did you make any progress on this issue? I'm seeing the same thing here and not quite sure how to solve it. For testing you can comment out the code that sets this parameter, but then there are other failures in the test cases that I suspect could be related to this parameter missing.

mkasenberg, 2024-03-20:
The `BadPixitException` was created to highlight that a PIXIT does not have any effect, either because of a typo in its name, wrong update value or the PIXIT does not exist anymore.

So after updating the BlueZ PTS workspace you have to remove not existing PIXITs from the autopts/ptsprojects/bluez/gap.py too.

It seems that the `TSPX_using_private_device_address` pixit has been removed/replaced by the `TSPX_using_public_device_address` in recent PTS versions.

closed 2024-04-16 by rlorenzoleal