auto-pts/auto-pts #1182: The wid blocks the PTS to send HCI to PTS dongle

MarkWangChinese, 2024-06-06:
I am trying to enable the HFP PTS test case in my side. In some cases. When one wid is triggered, one BTP cmd is sent to IUT, then wait the response of the BTP cmd in the wid function. When waiting for the response, the PTS doesn't send expected HCI cmd/response to the PTS dongle because the wid (for example: `hdl_wid_49(params: WIDParams)`) function is not returned. If the wid return quickly, I can see the HCI cmd/response is sent to PTS dongle quickly. Is it an issue?

piotrnarajowski, 2024-07-01:
Hello,
could you provide log files?

sjanc, 2024-09-11:
Hi,

Yes, we seen this behavior in PTS on few occasions (where PTS expect IUT action *after* WID is clicked).
 To handle this you may use post_wid functionality.

Example use:  https://github.com/auto-pts/auto-pts/blob/master/autopts/ptsprojects/zephyr/gatt.py#L329