auto-pts/auto-pts #1212: hid and related libraries not installed with autoptsserver_requirements.txt

MicImbriani, 2024-07-30:
If the steps in the README are followed, the workspace does not get set up correctly.
In fact, after running the ` python.exe -m pip install --user -r autoptsserver_requirements.txt`, all the dependencies should be met. However, when running the autoptsserver, the following Import Error is shown:

`ImportError: No module named hid`

And after pip installing hid:

`ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll`.

piotrnarajowski, 2024-08-01:
Hello,
thanks for letting know. autoptsserver_requirements.txt and autoptclient_requirements.txt will be updated. Just make sure you don't have installed both hid and hidapi module - the former might be interrupting loading hidapi. Let me know if you encounter any other issues.

MicImbriani, 2024-08-12:
I can see that the files have been updated by [PR#1214](https://github.com/auto-pts/auto-pts/pull/1214). 
Closing this issue.

closed 2024-08-12 by MicImbriani