auto-pts/auto-pts #1269: Request: Add suite result report

Thalley, 2024-10-07:
The reports uploaded to https://github.com/zephyrproject-rtos/bluetooth-qualification currently contains various reports, but it's non-trivial to see which suites (profiles, services, etc.) that are passing all currently enabled tests, and thus an overview of what can be considered qualify-able. 

It would good to have a report in some format that would show something like 

```
Suite   Total   Pass   Fail   Pass Rate
AICS    19      19     0      100 %
ASCS    56      56     0      100 %
CSIP    17      13     4      76 % 
```

This report should exist as its own downloadable file, but could also be embedded into the `README.md` to give a quick overview of each suite when looking at the https://github.com/zephyrproject-rtos/bluetooth-qualification repo.

piotrnarajowski, 2025-01-14:
@sjanc https://github.com/auto-pts/auto-pts/pull/1343 implements email summary

sjanc, 2025-02-18:
should be fixed now

closed 2025-02-18 by sjanc