auto-pts/auto-pts #1280: Fatal error when running `./autoptsclient-zephyr.py --help`

Thalley, 2024-10-23:
When running `autoptsclient-zephyr.py --help` the following error occurs:

```
$ ./autoptsclient-zephyr.py --help
usage: autoptsclient-zephyr.py [-h] [-i IP_ADDR [IP_ADDR ...]] [-l LOCAL_ADDR [LOCAL_ADDR ...]] [-a BD_ADDR] [-d] [-c TEST_CASES [TEST_CASES ...]] [-e EXCLUDED [EXCLUDED ...]] [--test_case_limit [TEST_CASE_LIMIT]] [-r RETRY] [--stress_test] [-S SRV_PORT [SRV_PORT ...]] [-C CLI_PORT [CLI_PORT ...]] [--recovery]
                               [--not_recover NOT_RECOVER [NOT_RECOVER ...]] [--superguard MINUTES] [--ykush YKUSH_PORT] [--pylink_reset] [--qemu_bin QEMU_BIN] [--hci HCI] [-t TTY_FILE] [-j DEBUGGER_SNR]
                               [-b {board_ext,nrf52,nrf52_wsl,nrf53,nrf53_appcore,nrf53_audio,nrf54h,nrf54l,nrf5x,nxp_evkbmimxrt1170,nxp_evkmimxrt1060,nxp_kw45b41zevk,reel_board}] [--btmon] [--device_core DEVICE_CORE] [--rtt-log] [--gdb]
                               [workspace] [kernel_image]

PTS automation client

positional arguments:
  workspace             Path to PTS workspace file to use for testing. It should have pqw6 extension. The file should be located on the machine, where automation server is running.
  kernel_image          OS kernel image to be used for testing,e.g. elf file for qemu, exe for native.

options:
  -h, --help            show this help message and exit
  -i IP_ADDR [IP_ADDR ...], --ip_addr IP_ADDR [IP_ADDR ...]
                        IP address of the PTS automation servers
  -l LOCAL_ADDR [LOCAL_ADDR ...], --local_addr LOCAL_ADDR [LOCAL_ADDR ...]
                        Local IP address of PTS automation client
  -a BD_ADDR, --bd-addr BD_ADDR
                        Bluetooth device address of the IUT
  -d, --debug-logs      Enable the PTS maximum logging. Equivalent to running test case in PTS GUI using 'Run (Debug Logs)'
  -c TEST_CASES [TEST_CASES ...], --test-cases TEST_CASES [TEST_CASES ...]
                        Names of test cases to run. Groups of test cases can be specified by profile names.Option can be used multiple times.
  -e EXCLUDED [EXCLUDED ...], --excluded EXCLUDED [EXCLUDED ...]
                        Names of test cases to exclude. Groups of test cases can be specified by profile names
  --test_case_limit [TEST_CASE_LIMIT]
                        Limit of test cases to run
  -r RETRY, --retry RETRY
                        Repeat test if failed. Parameter specifies maximum repeat count per test
  --stress_test         Repeat every test even if previous result was PASS
  -S SRV_PORT [SRV_PORT ...], --srv_port SRV_PORT [SRV_PORT ...]
                        Specify the server port number
  -C CLI_PORT [CLI_PORT ...], --cli_port CLI_PORT [CLI_PORT ...]
                        Specify the client port number
  --recovery            Specify if autoptsclient should try to recover itself after wrong status.
  --not_recover NOT_RECOVER [NOT_RECOVER ...]
                        Specify at which statuses autoptsclient should try to recover itself.
  --superguard MINUTES  Specify amount of time in minutes, after which super guard will blindly trigger recovery steps.
  --ykush YKUSH_PORT    Specify ykush downstream port number, so on BTP TIMEOUT the iut device could be powered off and on.
  --pylink_reset        Use pylink reset.
  --qemu_bin QEMU_BIN
  --hci HCI             Specify the number of the HCI controller(currently only used under native posix)
  -t TTY_FILE, --tty-file TTY_FILE
                        If TTY(or COM) is specified, BTP communication with OS running on hardware will be done over this TTY. Hence, QEMU will not be used.
  -j DEBUGGER_SNR, --jlink DEBUGGER_SNR
                        Specify jlink serial number manually.
  -b {board_ext,nrf52,nrf52_wsl,nrf53,nrf53_appcore,nrf53_audio,nrf54h,nrf54l,nrf5x,nxp_evkbmimxrt1170,nxp_evkmimxrt1060,nxp_kw45b41zevk,reel_board}, --board {board_ext,nrf52,nrf52_wsl,nrf53,nrf53_appcore,nrf53_audio,nrf54h,nrf54l,nrf5x,nxp_evkbmimxrt1170,nxp_evkmimxrt1060,nxp_kw45b41zevk,reel_board}
                        Used DUT board. This option is used to select DUT reset command that is run before each test case. If board is not specified DUT will not be reset.
  --btmon               Capture iut btsnoop logs from device over RTTand catch them with btmon. Requires rtt supporton IUT.
  --device_core DEVICE_CORE
                        Specify the device core for JLink related features, e.g. BTMON or RTT logging.
  --rtt-log             Capture iut logs from device over RTT. Requires rtt support on IUT.
  --gdb                 Skip board resets to avoid gdb server disconnection.
ERROR:root:0
Traceback (most recent call last):
  File "/home/thalley/dev/nordic/auto-pts/autopts/client.py", line 1363, in start
    return self.main(args)
           ^^^^^^^^^^^^^^^
  File "/home/thalley/dev/nordic/auto-pts/autopts/client.py", line 1376, in main
    errmsg = self.parse_config_and_args(_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thalley/dev/nordic/auto-pts/autopts/client.py", line 1332, in parse_config_and_args
    self.args, errmsg = self.arg_parser.parse(args_namespace)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thalley/dev/nordic/auto-pts/cliparser.py", line 268, in parse
    args = self.parse_args(None, arg_ns)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 1895, in parse_args
    args, argv = self.parse_known_args(args, namespace)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 1931, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 2168, in _parse_known_args
    start_index = consume_optional(start_index)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 2099, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.12/argparse.py", line 2006, in take_action
    action(self, namespace, argument_values, option_string)
  File "/usr/lib/python3.12/argparse.py", line 1144, in __call__
    parser.exit()
  File "/usr/lib/python3.12/argparse.py", line 2659, in exit
    _sys.exit(status)
SystemExit: 0
```

sjanc, 2024-10-24:
this seems to be python version (or more general environment) depended as I also see this on one box (linux) but not on the other (windows)

note:  help printed before traceback is complete :)

Thalley, 2024-10-24:
> this seems to be python version (or more general environment) depended as I also see this on one box (linux) but not on the other (windows)
> 
> note: help printed before traceback is complete :)

Updated the issue with the python version :) It sounds weird that this is related to the python version. Have you tried this with a clean repo? Looks like it's expecting some directory being somewhere where it's not by default.

sjanc, 2024-10-24:
ok, I stand corrected, on other box it simply revert order,  trace is printed first, then help message so I simply miss it :)

piotrnarajowski, 2024-11-14:
Fixed https://github.com/auto-pts/auto-pts/pull/1300

closed 2024-11-14 by Thalley