auto-pts/auto-pts #1283: Add symlink to most recent logs for the autopts client

Thalley, 2024-10-24:
When running the autopts client a new directory is created with the timestamp of the test run. 

For debugging purposes having a symlink to the most recent directory would make it easier to always have the most recent logs available in editors, without having to open the new files.

sjanc, 2025-01-09:
This is not only admin rights problem, but some filesystems (FAT, exFAT) doesn't support symlinks at all.

I believe this can be easily wrapped in shell by user after test run,  eg:
```rm -f latest; ln -s `ls -At |head -n 1` latest```

Thalley, 2025-01-09:
> This is not only admin rights problem, but some filesystems (FAT, exFAT) doesn't support symlinks at all.
> 
> I believe this can be easily wrapped in shell by user after test run, eg: `` rm -f latest; ln -s `ls -At |head -n 1` latest ``

How about
```
if fs_supports_symlinks(): 
    create_symlinks()
```

:D 

`twister` in Zephyr does basically what is requested here: The `twister-out` directory always contains the latest results, and older results are named `twister-out.1`, `twister-out.2`, etc. 

It doesn't need to be synlinks; it can also just be a copy, as long as it is possible to keep a specific file open for the latest results.

sjanc, 2025-01-09:
this will not work on windows, you cannot rename or delete folder if other process has file open

that said, I guess this could be under option for this (which will fail on windows),   yet it is trivial to just invoke mentioned snipped after client run :)

Thalley, 2025-01-09:
> this will not work on windows, you cannot rename or delete folder if other process has file open
> 
> that said, I guess this could be under option for this (which will fail on windows), yet it is trivial to just invoke mentioned snipped after client run :)

How about all logs from the current run is store in a generically named folder, e.g. "latest" or whatever. That's where everything is logged etc., and once the run has completed, the content of "latest" is copied to new directory with the current naming scheme with the test case name and datetime? That should work on all file systems and for any user that has write permission.

sjanc, 2025-01-09:
as mentioned, on windows, if you open any log file in latest (when run is ongoing), it is not possible to rename that folder as long as file is opened

Thalley, 2025-01-09:
> as mentioned, on windows, if you open any log file in latest (when run is ongoing), it is not possible to rename that folder as long as file is opened

I'm not suggesting any renames:

1. Start test with everything logged to the "latest" directory
2. Once test has completed, copy everything from "latest" to "<TEST_CASE_NAME>_<DATE_TIME>"

No files will be modified or renamed in this approach, and once a new test starts, simply overwrite the content of logs in "latest"