auto-pts/auto-pts #1320: `get_test_cases` reorders list of test cases

Thalley, 2024-12-12:
If the autoptsclient is called with `-c TEST_B TEST_A` then after `get_test_cases` `self.args.test_cases` becomes `["TEST_A", "TEST_B"]`. 

If a developer is debugging a test case and needs to run some tests in a specific order, they should be able to control the order when using the autoptsclient.

mringwal, 2024-12-12:
Good point. However, the order of tests should not matter, otherwise, something else is wrong :)

Thalley, 2024-12-12:
> Good point. However, the order of tests should not matter, otherwise, something else is wrong :)

For sure - And something is in fact wrong, and I needed to reorder the test to prove that the order of tests mattered :D

Thalley, 2024-12-12:
See also https://discord.com/channels/720317445772017664/1316767993522421830/1316800022981120114