auto-pts/auto-pts #1339: CSIP/CL/SP/BV-07-C regression. Unstable

koffes, 2025-01-13:
Looking at the Zephyr results, there is some instability on this test.
From the logs:

```
...
+23437 ms
Send Event:              ATT!ATT_READ_RSP{
                           lcid=0x1205,
                           length=0x01,
                           value{
                             FieldName=DefaultValue,
                             FieldType=OctetString
                           }

                         }

+113609 ms
Verdict Description:     User clicked No or Cancel button.

+113609 ms
Preliminary Verdict:     INDCSV

+113609 ms
Indecisive:              Non-passing verdict returned from APICOM, see log for more details.
...
```
A 90 sec timeout is triggered, likely due to some popup/callback not being handled properly.
I ran this manually as well with:

```
python ./autoptsserver.py -S 65000 65002 65004 --dongle COM40 COM38 COM42
python autoptsclient-zephyr.py zephyr-master -t COM9 -b nrf53_audio -S 65000 65002 65004  -C 65001 65003 65005 -c CSIP/CL/SP/BV-07-C
```

I get 12 failures because of the above, 2 fails for other reasons and 6 passing.
@mkasenberg, seems you are the most seasoned in this area. All tips are appreciated.

koffes, 2025-01-13:
After talking to sjanc, this may well be more general, and related to PTS sending multiple MMIs (man machine interface) which is not yet handled properly.

Thalley, 2025-01-13:
> This particular issue will be resolved or at least partially resolved by #1341

If you think the PR fixes this issue, please add a closing keyword in the PR to automatically close this issue when the PR is merged

closed 2025-01-14 by sjanc