auto-pts/auto-pts #1360: PTS debug logs mode is not properly restored after "PTS Timeout"

sjanc, 2025-01-28:
If PTS Timeout occurs during run debug mode setting is not used when PTS is restarted.
eg

$ ./autoptsclient_bot.py -c GAP/SEC/SEM/BV-65-C --debug-mode --nb --retry=10 --stress_test
DEBUG:root:Got free rtt for device 683802616: COM32
2090254956576 Starting PTS: 127.0.0.1:65000 ...
(2090254956576) OK
2090255128832 Starting PTS: 127.0.0.1:65002 ...
(2090255128832) OK
2090255132576 Starting PTS: 127.0.0.1:65004 ...
(2090255132576) OK
Number of test cases to run: 1
1/1   GAP     GAP/SEC/SEM/BV-65-C   PASS           38.745
1/1   GAP     GAP/SEC/SEM/BV-65-C   PASS           36.057    #1
1/1   GAP     GAP/SEC/SEM/BV-65-C   PASS           35.367    #2
1/1   GAP     GAP/SEC/SEM/BV-65-C   PASS           36.199    #3
1/1   GAP     GAP/SEC/SEM/BV-65-C   PTS TIMEOUT    322.154    #4
1/1   GAP     GAP/SEC/SEM/BV-65-C   INDCSV         34.124    #5     <---  from here PTS debug logs are not produced
1/1   GAP     GAP/SEC/SEM/BV-65-C   PASS           36.882    #6

piotrnarajowski, 2025-02-21:
Cannot reproduce this, but your log sample contains --debug-mode argument which should be --debug-logs (probably typo?)
Maybe try setting enable_max_logs to True in your config

sjanc, 2025-02-26:
I'll double check

closed 2025-02-27 by sjanc