auto-pts/auto-pts #1424: RTT logging is not working on nRF54H

alwa-nordic, 2025-03-25:
RTT logging is not working on nRF54H. The problem is that there is no "device_core" corresponding to the 54H in the JLink DLL yet.

I can set the generic `device_core='CORTEX-M33'`, but I also need to be able to set the RTT block address. It works if I hardcode it as shown below, in `autopts.rtt.RTT.init_jlink`:

```python
RTT.jlink.rtt_start(block_address=0x000000002f013010)
```

**Expected solution**: An `rtt_control_block_address` parameter in `client.py`.

**Expected solution**: There should be an error reported when RTT logs are requested by the user, but are configured incorrectly.

alwa-nordic, 2025-03-25:
I'm still trying to orient myself around auto-pts, but I think since auto-pts also governs compilation, we'll might want to implement discovery of the RTT control block address from the Zephyr elf file.

I'm not sure how best to structure that. I think the right place might be in `autopts/ptsprojects/boards/nrf54h.py`. It could be made a return value from `build_and_flash`.

To get the address of the RTT control block we use `binutils`, like so:

```sh
addr2line -e build/tester/zephyr/zephyr.elf _SEGGER_RTT -a | head -1
```

The addresses of symbols can also be found in `build/tester/zephyr/zephyr.map`. `.map` is not really meant to be machine readable, but it could be used instead if we don't want to depend on `binutils` being on PATH.

sjanc, 2025-03-26:
is this something that changes on each build?  I'm bit reluctant with 'hackish' autodetection (we have some detection for com ports and nrf serial detection but it tends to break from time to time causing user confusion) if this is fixed and can be configured via eg. config.py,   or explicit config that autodetection should be used instead of provided address

alwa-nordic, 2025-03-26:
It looks like [the RTT address can be specified in the vector table](https://kb.segger.com/RTT#Auto-detection). I'll look into that.

alwa-nordic, 2025-03-26:
That was a red herring. The linked documentation is not entirely correct. The vector table does not have a standard memory address, so JLink does not know where it is in a generic Cortex-M33 device.

alwa-nordic, 2025-03-26:
> is this something that changes on each build?

Potentially. We could place it at a specific address with a linker script.