auto-pts/auto-pts #1464: Use of nrfjprog is deprecated

koffes, 2025-05-21:
In several files, there is a reference to 

`return f'nrfjprog -r -s {iutctl.debugger_snr}'`. The issue is that nrfjprog is deprecated (See https://www.nordicsemi.com/Products/Development-tools/nRF-Command-Line-Tools)

It should rather use something like:

`nrfutil device reset --serial-number {iutctl.debugger_snr}`

koffes, 2025-05-21:
The latter is preferable, as that abstracts the call and should work both with nrfutil and nrfjprog as runners.

Thalley, 2025-05-21:
`west` is of course only useful when flashing Zephyr. If the nRF boards are used for non-Zephyr builds, you'd have to use `nrfutil`

m-alperen-sener, 2025-11-28:
I have tried to remove more from the boards; but I am not sure about the mynewt usage, looks like there using nrfjprog also for flashing, so I did not touch nordic_pca_XXXX.py

https://github.com/auto-pts/auto-pts/pull/1640