auto-pts/auto-pts #1511: forcing overlay-bt_ll_sw_split.conf file for every build causing failure for mesh

m-alperen-sener, 2025-07-09:
I am not sure why this overlay-bt_ll_sw_split.conf file is overlay by dafault to every nrf5x devices; but it overwrites a lot of mesh configs; and mesh currently run only on single core devices.

Change introduces here https://github.com/auto-pts/auto-pts/commit/6ef44482de2974a6aecb7195f4c560c22e207830

Thalley, 2025-07-14:
As we've discussed in the weekly (and probably also on Discord), I'm in favor of the AutoPTS client not providing any specific configurations when building. 

Ideally the BT Tester application in Zephyr should just build with `west build -p <board> --sysbuild` and then it would build with whatever is needed for the tests; no configuration files should be necessary from AutoPTS. 

At the moment that's impossible due to e.g. some tests requiring privacy to be enabled, and some requiring privacy to be disabled which is a Kconfig option, and not a runtime option, in Zephyr, so as of today we kind of need to do something. 

But the `overlay-bt_ll_sw_split.conf` should IMO definitely not be added by AutoPTS as that basically prevents building on nRF platforms for other controllers than the Zephyr controller.

sjanc, 2025-07-14:
I think that any zephyr based platform that differs from upstream should have own build_and_flash() implementation

m-alperen-sener, 2025-07-15:
I have tried to read discord chat but did not see a detailed discussion about this, or any outcome; maybe I should joined the zephyr weekly. But anyway there is no one way of doing it at the end right; it will be combination of cmake, testcase.yml and some Kconfig magic.

Thalley, 2025-07-15:
> I have tried to read discord chat but did not see a detailed discussion about this, or any outcome; maybe I should joined the zephyr weekly. But anyway there is no one way of doing it at the end right; it will be combination of cmake, testcase.yml and some Kconfig magic.

The testcase.yml approach can do all of it without involving Cmake or any Kconfig "magic" IMO. It will however require that the user builds using `west -T` and specifying the correct configuration, so that will be something that needs to be documented as well as applied to AutoPTS in some way

m-alperen-sener, 2025-07-15:
AFAIS Kconfig is still needed in some cases: for example mesh uses persistent storage; and it requires selecting NVS or ZMS depending on the target; it will be nasty if we put that into yml file but it would be nice if we factor some common changes into tester Kconfig instead, IMO. I am totally onboard with west build with test_items because it will introduce build only test coverage for every configuration which we don't have now and currently available with weekly PTS run; a huge latency to get regression.